SJTU-ViSYS/M2DGR #4: xsens mti的IMU数据似乎没有提供

wangyu-060070, 2021-12-30:


sjtuyinjie, 2021-12-30:
As mentioned in my paper, the IMU infomation of XSENS is used for generating GT pose. If you need 9 axis IMU information for LiDAR SLAM, use the topic /handsfree/imu instead.

wangyu-060070, 2021-12-30:
Figure 2 is mislabeled and there are two label-5. In addition, we calibrated the imu-camera extrinsic, the results show that the hander IMU is above the infrared

sjtuyinjie, 2021-12-30:
Sorry, the left one of the "5" labels should be "7". But the handsfree IMU is below the infrared camera as shown in the figure.

sjtuyinjie, 2021-12-30:
Maybe the IMU in the VI-sensor (Intel Realsense d435i) is above the infrared.

wangyu-060070, 2021-12-31:
I utilize the Handsfree imu data, and the result shows that the IMU is above the infra camera

从 Windows 版邮件发送

发件人: Yin Jie
发送时间: 2021年12月30日 22:09
收件人: SJTU-ViSYS/M2DGR
抄送: Barrier; Author
主题: Re: [SJTU-ViSYS/M2DGR] xsens mti的IMU数据似乎没有提供 (Issue #4)

Maybe the IMU in the VI-sensor (Intel Realsense d435i) is above the infrared.

sjtuyinjie, 2021-12-31:
Please describe the detailed process how you got the this result.

closed 2022-01-04 by sjtuyinjie