SJTU-ViSYS/M2DGR #9: Lack of Extrinsic Parameter from Mocap to Lidar

oym1994, 2022-01-18:
Hi, thanks for your great job!
I am going to use the room data to test my algorithm. As is mentioned in the paper, the groundtruth is provided by a Mocap. But I cant find the related extrinsic parameter of this sensor. Would you mind updating the calibration_results.txt as soon as possible to provide this necessary extrinsic parameter? Or can I get the extrinsic parameter  by calculating other extrinsic parameters?  
Thank you in advance!

sjtuyinjie, 2022-01-18:
As mentioned in the paper, the ground truth trajectories of Room and Roomdark sequences are obtained by Mocap. In fact, Mocap paremeters are not necessary, because when you evaluate your SLAM result, the "-a" command of EVO will align the GT with your trajectories. We do not have to specially calibrate the extrinsics between LiDAR and Mocap.

oym1994, 2022-01-19:
Got it! Actually we want to build the map with Room datasets with ground-truth poses and locate with the Roomdark datasets. In this case, the absolute ground-truth pose is necessary. But if the extrinsic parameter is not available, just ignore it. Anywaym thank you for your kind response!

sjtuyinjie, 2022-01-19:
OK. If any other questions, feel free to propose issues.

closed 2022-01-19 by sjtuyinjie

yuhui-xie, 2022-06-28:
> As mentioned in the paper, the ground truth trajectories of Room and Roomdark sequences are obtained by Mocap. In fact, Mocap paremeters are not necessary, because when you evaluate your SLAM [...]
 
It means that actually the EVO will estimate the extrinic in the alignment process? Like hand-eyes calibration between two trajectory?

LarryDong, 2024-07-19:
> It means that actually the EVO will estimate the extrinic in the alignment process? Like hand-eyes calibration between two trajectory?

No. EVO using Umeyama to align trajectories, but cannot estimate the extrinsics. 

For example, if the platform is rotating at the center of the Lidar frame, the trajectory of lidar is a simple point. But for Mocap, the trajectory is a circle. So these two trajectory cannot be aligned by evo anyway. 
That is to say, the extrinsics translation will effect the trajectory, but the extrinsics rotation can aligned by evo's alignment (the platform always have the same rotation).
However, if the lidar & Mocap translation is small, or the rotation duration is not long, the effect is not obvious if the whole trajectory is relative long. But it still introduces some error. 

As a dataset for evaluating the algorithms, I think this issue should be avoid.