SJTU-ViSYS/M2DGR #10: The quaternion of the ground-truth is all 0 and wrong GT link

oym1994, 2022-01-21:
Hi, Thanks for your great job!
I found that the quaternion  of the ground truth which is provided by the Leica is all 0. Is it for Leica only provides 3 DOF position estimation?
Another problem is that some GT link is wrong. For example the GT link of hall_03 will guild you to the hall_02. I suggest to check those links again.
Thanks and wait for your kind response!

sjtuyinjie, 2022-01-22:
Good question. Leica can only track 3D position rather than 6D position and pose. 
As for GT problem, sorry for my oversight. I will double check it right now.

sjtuyinjie, 2022-01-22:
I have re-uploaded GT file of Hall03 here in the link https://github.com/SJTU-ViSYS/M2DGR/blob/main/hall_03.txt

closed 2022-01-22 by sjtuyinjie

oym1994, 2022-01-24:
Another problem is that the ground-truth website of hall_04 is linked to room_dark-04, lift_03 and lift_04 are linked to hall_05.  please check it!

sjtuyinjie, 2022-01-24:
It's okay now. Contact me if any new questions.