SJTU-ViSYS/M2DGR #17: GNSS data of M2DGR

Lchhehehe, 2022-04-02:
1)The M2DGR literature points out that there are two GNSS receivers' data, but I see that the schematic diagram has only one antenna. Do the two receivers share the same antenna, one of which records the original observation and the other records the real trajectory?
2)Can you provide documents such as the format description of topic, or where you can find such documents? It seems that the topic description file on the ROS official website does not contain the description of the raw observations of GNSS.
3)The operation of converting rosbag (street_02.bag) into GNSS raw observation data through the ublox2rinex program failed, and only the header file of RINEX was generated.

Lchhehehe, 2022-04-02:
[image]

sjtuyinjie, 2022-04-04:
(1). Yes, GNSS signals are received by  a single antenna and shared to two GNSS receivers by a power splitter.
(2). You can look up to GNSS raw observation.
(3).You may need to  slightly modify the configuration like changing the file path. I am sure the toolkit can successfully convert Rosbag to GNSS raw observation data because I have done it before.

closed 2022-04-04 by sjtuyinjie