SJTU-ViSYS/M2DGR #22: 关于数据集下载方式的一个请求

liang-jinli, 2022-06-07:
您好，

首先非常感谢贵实验室对SLAM领域做出的贡献，我计划用该数据集做一些相关实验来尝试解决一些问题。

但是在下载数据集时，通过OneDrive方式的下载速度有些慢并经常中途下载失败。

这里想请问你们能否提供对国内用户更友好的下载方式，比如百度网盘？

希望得到您的回复~

sjtuyinjie, 2022-06-08:
百度网盘限制文件大小为20g，无法满足需求
阿里云盘无法上传bag格式的文件
google/dropbox国内无法登录
而onedrive在国内的校园网下载速度大概为10MB/s，速度已经是最快的了。

closed 2022-06-08 by sjtuyinjie

liang-jinli, 2022-06-08:
好的，多谢回复~

再次感谢贵实验室做出的贡献！

DreamWaterFound, 2022-06-28:
我觉得有这么几种解决方法：
- 奶牛快传或者文叔叔，这两个网站针对影视行业的大文件传输设计，单个文件大小限制我记不太清楚了，付费服务比较贵；
- 利用阿里云腾讯云的对象存储服务，北研院（作者应该知道我在说什么）网站的数据集就是这种托管方式，文件大小可以说几乎不受限制，但容量费用+下行流量费用也不是个小数字；
- 对于教育网用户（学生），使用PT站做种上传也不错；对于校内网络收费的同学来说，走IPV6下载数据集还可以不计费。但是如果没人做种，或者用的高校PT站暂停服务了（比如前一阵的蒲公英）这个就断种了，参考Brno数据集最近的[问题](https://github.com/Robotics-BUT/Brno-Urban-Dataset/issues/4)，也不知道修复好了没有。同时对于非教育网用户，也就是社会上的小伙伴们就没法用了；
- 最合适的办法是对原始数据集进行分卷压缩，同时提供文档告知用户下载后如何将每个分卷合成一个大的压缩文件，以及对比MD5、SHA1等校验码确保用户拼接后的文件和原文件一致准确。这样上传到什么地方都可以。缺点就是，作者自己需要开展的工作量不小。

66695, 2022-06-29:
请问onedrive下载的连接在哪呢