SJTU-ViSYS/M2DGR #24: imu的标定

Yang-lin-09, 2022-06-09:
请问一下，你们的IMU是怎么进行标定的？这个型号的IMU，我也使用imu_utils进行标定，不过标定出来的结果是nan。所以，想请问一下你们是直接使用ros获取的imu bag进行标定的嘛？还是进行了一些特殊的处理？非常感谢。

Yang-lin-09, 2022-06-09:
%YAML:1.0
---
type: IMU
name: HFI_a9
Gyr:
   unit: " rad/s"
   avg-axis:
      gyr_n: .Nan
      gyr_w: .Nan
   x-axis:
      gyr_n: .Nan
      gyr_w: .Nan
   y-axis:
      gyr_n: .Nan
      gyr_w: .Nan
   z-axis:
      gyr_n: .Nan
      gyr_w: .Nan
Acc:
   unit: " m/s^2"
   avg-axis:
      acc_n: 8.1632539596839658e+36
      acc_w: 1000.
   x-axis:
      acc_n: 3.3180421057132749e+36
      acc_w: 1000.
   y-axis:
      acc_n: 9.7918439984517541e+36
      acc_w: 1000.
   z-axis:
      acc_n: 1.1379875774886867e+37
      acc_w: 1000.

这是我标定的结果

sjtuyinjie, 2022-06-11:
录制bag，静止2小时以上，直接跑bag即可

closed 2022-06-11 by sjtuyinjie