SJTU-ViSYS/M2DGR #31: Questions about Lidar-IIMU calib

TongxingJin, 2022-06-26:
Hi, thks for your nice work. For the extrinsic between Lidar and IMU, you refer to [Lidar_IMU_Calib], did it works well for ground vehicles without sufficient excitement? And in the livestream tonight, you also said you did measurements manually, so what information did you adopt from Lidar_IMU_Calib?

sjtuyinjie, 2022-06-27:
I once calibrated the extrinsic between Lidar and IMU using this work which  gave a bad result. I used to publish this result in the calibration_result.txt file, but I replace it with a more accurate one later.

closed 2022-06-27 by sjtuyinjie

TongxingJin, 2022-06-27:
> I once calibrated the extrinsic between Lidar and IMU using this work which gave a bad result. I used to publish this result in the calibration_result.txt file, but I replace it with a more accurate one later.

And then how you get a more accurate one? by manual measurement?

sjtuyinjie, 2022-06-27:
Right. This is the common method when calibration algorithms fail.