SJTU-ViSYS/M2DGR #41: 您好，想咨询一下关于GNSS发布话题的相关问题

MoonSheepJx, 2022-08-30:
您好，首先非常感谢您的工作！
关于GNSS发布的相关话题，数目较多，但是对其说明的资料较少，想了解一下GNSS相关话题的意义或用途，比如我该订阅哪个消息以获得经纬信息等，其他话题大概是关于什么的。
希望您能提供一些参考，再次非常感谢您的工作，数据集很棒！

sjtuyinjie, 2022-08-30:
/ublox/fix 是经纬度。其他topic是星历等信息，具体内容你可以去https://github.com/KumarRobotics/ublox看看。M2DGR录制的是m8t的数据，目前大家都用f9p了，m8t已经有点过时了

closed 2022-08-30 by sjtuyinjie

sjtuyinjie, 2022-08-31:
我录制了带有f9p原始信息的新数据。你可以去测试一下：https://github.com/sjtuyinjie/SJTU_GVI

MoonSheepJx, 2022-08-31:
好的，稍后我会使用这个去测试一下，多谢您的回复与工作！

wjf1997, 2022-09-22:
> 好的，稍后我会使用这个去测试一下，多谢您的回复与工作！

你好，我下载数据集的时候容易中断，请问有什么方法可以完整下载数据集吗

MoonSheepJx, 2022-09-23:
> > 好的，稍后我会使用这个去测试一下，多谢您的回复与工作！
> 
> 你好，我下载数据集的时候容易中断，请问有什么方法可以完整下载数据集吗

多下载几次就好了，下载的时候确实容易中断，不过多点几次就可以成功了，我是这样的

Lchhehehe, 2022-09-28:
您好，数据中提供的GNSS相关数据，里面是不是没有卫星的星历信息？我看topic里面好像没有

sjtuyinjie, 2022-09-28:
> 您好，数据中提供的GNSS相关数据，里面是不是没有卫星的星历信息？我看topic里面好像没有

试试https://github.com/sjtuyinjie/SJTU_GVI，这个里面有。记得给个star哈~

Lchhehehe, 2022-10-18:
感谢感谢，请问GNSS采集的数据有没有多频多系统的数据？

Lchhehehe, 2022-11-10:
还有一个问题，怎么将数据中的周（week）和周内秒（sow）与系统的时标(stamp)对应起来？

huanhexiao, 2024-08-18:
> /ublox/fix [...]

请问/ublox/fix有使用地面基站作差分吗