SJTU-ViSYS/M2DGR #52: 关于LIO-SAM的建图问题

ZhangZh3ng, 2022-10-28:
作者你好，首先非常感谢你们慷慨的开源M2DGR数据集！

最近我们使用此数据集对LIO-SAM进行了测试（使用你们所提供的配置参数）。在我们已下载到的多数M2DGR轨迹上，最终建图效果并不理想。与在其他数据集上LIO-SAM的建图结果相比差别比较大。
例如：
在street_08上的建图效果：
[image]
在LIO-SAM所提供的数据上的建图效果：
[image]
可以明显的发现前者建出的地图有些糊，而后者就清晰许多。
由于你们提供了LIO-SAM的配置文件，想必也在LIO-SAM上进行过测试，因此我们非常好奇你们的测试结果。请问你们如何看待此现象，有没有能改进建图结果的建议，谢谢！

sjtuyinjie, 2022-10-29:
我在自己的笔记本测试的时候Liosam的建图效果还挺好的。从你对比的两张图来看，M2DGR的数据是在室外不规则的街道上，有许多不规则的树叶等信息，所以点云的连贯性看上去没那么好。第二张图似乎是建筑物内，点云更规则一点。此外，建图效果和你的笔记本的性能也有一定关系。如果是点云出现大量重影的话，那么是定位端出现了问题

closed 2022-10-29 by sjtuyinjie

ZhangZh3ng, 2022-10-30:
> 我在自己的笔记本测试的时候Liosam的建图效果还挺好的。从你对比的两张图来看，M2DGR的数据是在室外不规则的街道上，有许多不规则的树叶等信息，所以点云的连贯性看上去没那么好。第二张图似乎是建筑物内，点云更规则一点。此外，建图效果和你的笔记本的性能也有一定关系。如果是点云出现大量重影的话，那么是定位端出现了问题

我们上传了一些测试结果（包括四个M2DGR轨迹和一个LIO-SAM自带的轨迹，它们都是在同一台电脑上得到的），能否请你将其与你们的结果进行对比呢？谢谢!
https://drive.google.com/file/d/1bhGpCxgoODGcdRS3W-BFLhW7VDr_Z8PE/view?usp=sharing

MoonSheepJx, 2022-11-02:
您好，我也正在使用LIO-SAM测试M2DGR，我测试的结果也表示建图效果不好，但定位效果还可以，请问是不是我的配置文件有些问题
配置文件如下:
[image]
[image]
效果如下：
[image]

能否提供你的LIO-SAM配置文件来供参考，非常感谢

MoonSheepJx, 2022-11-02:
您好，我也正在使用LIO-SAM建图，之前在配置文件中忘记修改雷达的线数了，建图效果非常模糊，不知道您是不是这个问题
模糊图如下
[image]
修改后如下
[image]
不知道这是不是正常的，从建图看还可以的，整体轨迹也对的上，但是局部定位有些抖动，请问这大概是算法的事情还是没有配置好的问题
我的配置文件如下
[image]
[image]
（暂时没有使用GNSS数据，测试的数据集为street_03）

ZhangZh3ng, 2022-11-02:
感谢你的提醒，我检查配置文件确认了N_SCAN为32，street_03我还没下载到，之后下载到后我会进行测试。

ZhangZh3ng, 2022-11-05:
经过我们的测试发现，问题的原因是rviz中点云的显示模式造成的，将Style由“Points”改成“Flat Squares”（size = 0.01）看起来就正常了
[image]

wjf1997, 2022-11-26:
> 作者你好，首先非常感谢你们慷慨的开源M2DGR数据集！
> 
> 最近我们使用此数据集对LIO-SAM进行了测试（使用你们所提供的配置参数）。在我们已下载到的多数M2DGR轨迹上，最终建图效果并不理想。与在其他数据集上LIO-SAM的建图结果相比差别比较大。 例如： 在street_08上的建图效果： [image] 在LIO-SAM所提供的数据上的建图效果： [image] 可以明显的发现前者建出的地图有些糊，而后者就清晰许多。 由于你们提供了LIO-SAM的配置文件，想必也在LIO-SAM上进行过测试，因此我们非常好奇你们的测试结果。请问你们如何看待此现象，有没有能改进建图

> 感谢你的提醒，我检查配置文件确认了N_SCAN为32，street_03我还没下载到，之后下载到后我会进行测试。

请问你是在国内环境下载的数据集吗？有没有啥好方法，下载一直中断

wjf1997, 2022-11-26:
> 您好，我也正在使用LIO-SAM建图，之前在配置文件中忘记修改雷达的线数了，建图效果非常模糊，不知道您是不是这个问题 模糊图如下 [image] 修改后如下 [image] 不知道这是不是正常的，从建图看还可以的，整体轨迹也对的上，但是局部定位有些抖动，请问这大概是算法的事情还是没有配置好的问题 我的配置文件如下 [image] [image] （暂时没有使用GNSS数据，测试的数据集为street_03）

请问你是在国内环境下载的数据集吗？请问下载一直中断怎么办呀

Lchhehehe, 2022-12-12:
我在用LIO-SAM处理数据时（street_02）正常速度播放数据包时会出现卡顿，进而导致程序跳变最后定位和建图失败，0.5倍速播放时不会出现这个问题，这是电脑处理能力的问题吗，我用的戴尔的台式机i9处理器

orange-cr, 2023-03-14:
> 您好，我也正在使用LIO-SAM建图，之前在配置文件中忘记修改雷达的线数了，建图效果非常模糊，不知道您是不是这个问题 模糊图如下 [image] 修改后如下 [image] 不知道这是不是正常的，从建图看还可以的，整体轨迹也对的上，但是局部定位有些抖动，请问这大概是算法的事情还是没有配置好的问题 我的配置文件如下 [image] [image] （暂时没有使用GNSS数据，测试的数据集为street_03）

orange-cr, 2023-03-14:
> 您好，我也正在使用LIO-SAM建图，之前在配置文件中忘记修改雷达的线数了，建图效果非常模糊，不知道您是不是这个问题 模糊图如下 [image] 修改后如下 [image] 不知道这是不是正常的，从建图看还可以的，整体轨迹也对的上，但是局部定位有些抖动，请问这大概是算法的事情还是没有配置好的问题 我的配置文件如下 [image] [image] （暂时没有使用GNSS数据，测试的数据集为street_03）

请问定位抖动的问题解决了么，可以分享一下大概的思路和解决方法吗