SJTU-ViSYS/M2DGR #65: 数据集中图像掉帧、卡顿问题

Tianci-Wen, 2023-02-03:
您好！我运行了数据集gate_01、street_06、room_dark_01这三个序列，发现无论是否运行slam算法，图像信息(/camera/color/image_raw/compressed)时不时的会出现卡顿现象，特别是room_dark_01这个序列尤为严重。
以数据集gate_01为例，总时长172.547653s，相机频率为15Hz，应有2588个图像消息，但是实际上bag里只有2490个消息，少了近100帧。
[image]

我电脑cpu型号为11th Gen Intel® Core™ i7-11800H ，不知道是不是我电脑配置的问题，但是之前运行其他数据集时没出现这种情况。不知道您有什么解决的思路，非常感谢您！

sjtuyinjie, 2023-02-07:
卡顿和掉帧是两个问题。卡顿主要是cpu计算性能不足引起，如果你想保证结果的话那就半速播放bag，例如rosbag play xxx.bag -r 0.5。掉帧是realsense d435I本身的硬件问题，15Hz并不完全精准，有时可能是14出头

closed 2023-02-07 by sjtuyinjie

zhh2005757, 2024-12-14:
> 卡顿和掉帧是两个问题。卡顿主要是cpu计算性能不足引起，如果你想保证结果的话那就半速播放bag，例如rosbag play xxx.bag -r

我也遇到了图像卡顿的问题，播放其他的d435i数据就没有图像卡顿，貌似不是cpu性能的问题。