SJTU-ViSYS/M2DGR #66: How to convert the thermal infrared image in 8UC3 format to the image in mono8 image available in SLAM

504115681, 2023-02-07:
Hello, this is a good job! Now I have some problems. I don't know how to convert the thermal infrared image in 8UC3 format to mono8 image available in SLAM. Thank you for any help!

sjtuyinjie, 2023-02-07:
Here is a demo code converting 8UC1 to mono8. You can take a look:

cv::Mat getImageFromMsg(const sensor_msgs::ImageConstPtr &img_msg)
{
  cv_bridge::CvImageConstPtr ptr;
  if (img_msg->encoding == "8UC1")
  {
    sensor_msgs::Image img;
    img.header = img_msg->header;
    img.height = img_msg->height;
    img.width = img_msg->width;
    img.is_bigendian = img_msg->is_bigendian;
    img.step = img_msg->step;
    img.data = img_msg->data;
    img.encoding = "mono8";
    ptr = cv_bridge::toCvCopy(img, sensor_msgs::image_encodings::MONO8);
  }
  else
    ptr = cv_bridge::toCvCopy(img_msg, sensor_msgs::image_encodings::MONO8);

  cv::Mat img = ptr->image.clone();

  return img;
}

try replacing "8UC1" with "8UC3"

closed 2023-02-07 by sjtuyinjie

504115681, 2023-02-09:
> Here is a demo code converting 8UC1 to mono8. You can take a look:
>
> cv::Mat getImageFromMsg(const sensor_msgs::ImageConstPtr &img_msg) { cv_bridge::CvImageConstPtr ptr; if (img_msg->encoding == [...]

I added the above code in VINS-Mono, and it ran successfully. However, when I looked at the original thermal image and the converted image, I found that the original image was elongated after being cropped. Is there a problem in the conversion process? Thanks again!
[image]