SJTU-ViSYS/M2DGR #88: Handsfree A9 IMU与雷达的时间同步问题

YWL0720, 2023-10-09:
感谢你们的工作！有一个疑问想要请教一下。论文中提到了数据集所用传感器之间的时间同步方式，是采用同一系统时间。我了解到Handsfree A9所提供的接口有限，想请问一下，Handsfree A9和激光雷达等传感器具体是如何进行时间同步的，是ptp之类的方法么？目前我也遇到了这方面的需求，希望能得到一些建议。感谢您的回复！

sjtuyinjie, 2024-02-08:
事实上这一块处理得不够严谨，确实应该使用ptp等方式同步。

closed 2024-02-08 by sjtuyinjie

Nostalgic-cccc, 2026-10-04:
Thank you for the clarification about Handsfree A9 synchronization. Could you help clarify the separate ground-truth timestamp/export chain for room_01 (Vicon) and hall_02 (Leica)? The outdoor Xsens/MTManager discussion in issue 77 appears to cover a different chain.

For each recording, could you provide the raw GT format and export script/version, what instant the published GT timestamp represents (measurement, frame time, host receipt, or another event), the originating clock and mapping to the ROS/LiDAR clock, any offset already applied with its sign convention, and any synchronization log or measured residual uncertainty? For LiDAR, does the point-cloud header mark scan start, scan end, or another reference instant, and which driver/version established that definition?

The linked image-export scripts seem to preserve image header timestamps but do not describe Vicon/Leica GT export. If the original export or calibration records are unavailable, could you explicitly identify which time mappings were assumed rather than independently measured? This would let us report the accuracy evidence without treating timestamp formatting or recording latency as a calibrated offset. Thank you.