SJTU-ViSYS/M2DGR #101: How to translate your ground truth from  ECEF frame to ENU frame?

HowieHepburn, 2024-05-22:
First and foremost, I would like to extend my gratitude to you and your team for providing such an outstanding and comprehensive dataset. The data has been immensely helpful for our research work.

I am currently in the process of writing a paper and need to showcase trajectories. To ensure a unified perspective when using the evo tool, I utilize commands like plot_mode=xy. However, I have encountered an issue where the provided ground truth data is not in the ENU (East-North-Up) frame, with the Z-axis pointing upwards, but rather in the ECEF (Earth-Centered, Earth-Fixed) frame. This discrepancy has caused some difficulties in my work.

Therefore, I would like to inquire about the procedure to convert the ground truth data from the ECEF coordinate system to the ENU coordinate system. Any guidance or recommendations you could provide would be greatly appreciated.

Thank you once again for your valuable data and support. I look forward to your response.

HowieHepburn, 2024-05-23:
Well,I think I have solved this problem but still need more tests.

The scripts works well.
[image]
[image]
[image]
[image]

import numpy as np
import argparse

def ecef_to_enu(lat_ref, lon_ref, alt_ref, x, y, z):
    # WGS84 ellipsoid constants
    a = 6378137.0  # semi-major axis in meters
    e = 8.1819190842622e-2  # first eccentricity

    # Compute prime vertical radius of curvature
    N = a / np.sqrt(1 - e**2 * np.sin(np.radians(lat_ref))**2)

    # Compute ECEF coordinates of the reference point
    x_r = (N + alt_ref) * np.cos(np.radians(lat_ref)) * np.cos(np.radians(lon_ref))
    y_r = (N + alt_ref) * np.cos(np.radians(lat_ref)) * np.sin(np.radians(lon_ref))
    z_r = (N * (1 - e**2) + alt_ref) * np.sin(np.radians(lat_ref))

    # Compute the difference between the point and the reference point
    dx = x - x_r
    dy = y - y_r
    dz = z - z_r

    # Compute the rotation matrix
    R = np.array([
        [-np.sin(np.radians(lon_ref)), np.cos(np.radians(lon_ref)), 0],
        [-np.sin(np.radians(lat_ref)) * np.cos(np.radians(lon_ref)), -np.sin(np.radians(lat_ref)) * np.sin(np.radians(lon_ref)), np.cos(np.radians(lat_ref))],
        [np.cos(np.radians(lat_ref)) * np.cos(np.radians(lon_ref)), np.cos(np.radians(lat_ref)) * np.sin(np.radians(lon_ref)), np.sin(np.radians(lat_ref))]
    ])

    # Apply the rotation matrix to the difference vector
    enu = R @ np.array([dx, dy, dz])
    
    return enu

def convert_tum_ecef_to_enu(input_file, output_file, lat_ref, lon_ref, alt_ref):
    with open(input_file, 'r') as f_in, open(output_file, 'w') as f_out:
        for line in f_in:
            if line.strip() and not line.startswith('#'):
                parts = line.strip().split()
                timestamp = parts[0]
                x_ecef = float(parts[1])
                y_ecef = float(parts[2])
                z_ecef = float(parts[3])
                qx = parts[4]
                qy = parts[5]
                qz = parts[6]
                qw = parts[7]
                
                enu = ecef_to_enu(lat_ref, lon_ref, alt_ref, x_ecef, y_ecef, z_ecef)
                f_out.write(f"{timestamp} {enu[0]:.6f} {enu[1]:.6f} {enu[2]:.6f} {qx} {qy} {qz} {qw}\n")

if __name__ == "__main__":
    parser = argparse.ArgumentParser(description="Convert ECEF coordinates in TUM format to ENU coordinates.")
    parser.add_argument("input_file", type=str, help="Path to the input TUM file with ECEF coordinates.")
    parser.add_argument("output_file", type=str, help="Path to the output TUM file with ENU coordinates.")
    parser.add_argument("lat_ref", type=float, help="Reference latitude in degrees.")
    parser.add_argument("lon_ref", type=float, help="Reference longitude in degrees.")
    parser.add_argument("alt_ref", type=float, help="Reference altitude in meters.")

    args = parser.parse_args()

    convert_tum_ecef_to_enu(args.input_file, args.output_file, args.lat_ref, args.lon_ref, args.alt_ref)

closed 2024-05-23 by HowieHepburn

Urtii, 2024-11-26:
Thanks, I also added a function to your code to get LLA referance from NavSatFix message of a .bag files /ublox/fix topic if anyone needs quicker solution.

just call

`./ecef_to_xyz_wrt_bag.py <input_ecef_file.txt> <output_file.txt> <referance_file.bag>`

```
#!/usr/bin/env python

import numpy as np
import argparse
import rosbag
from roslib import message
from sensor_msgs.msg import NavSatFix

def ecef_to_enu(lat_ref, lon_ref, alt_ref, x, y, z):
    # WGS84 ellipsoid constants
    a = 6378137.0  # semi-major axis in meters
    e = 8.1819190842622e-2  # first eccentricity

    # Compute prime vertical radius of curvature
    N = a / np.sqrt(1 - e**2 * np.sin(np.radians(lat_ref))**2)

    # Compute ECEF coordinates of the reference point
    x_r = (N + alt_ref) * np.cos(np.radians(lat_ref)) * np.cos(np.radians(lon_ref))
    y_r = (N + alt_ref) * np.cos(np.radians(lat_ref)) * np.sin(np.radians(lon_ref))
    z_r = (N * (1 - e**2) + alt_ref) * np.sin(np.radians(lat_ref))

    # Compute the difference between the point and the reference point
    dx = x - x_r
    dy = y - y_r
    dz = z - z_r

    # Compute the rotation matrix
    R = np.array([
        [-np.sin(np.radians(lon_ref)), np.cos(np.radians(lon_ref)), 0],
        [-np.sin(np.radians(lat_ref)) * np.cos(np.radians(lon_ref)), -np.sin(np.radians(lat_ref)) * np.sin(np.radians(lon_ref)), np.cos(np.radians(lat_ref))],
        [np.cos(np.radians(lat_ref)) * np.cos(np.radians(lon_ref)), np.cos(np.radians(lat_ref)) * np.sin(np.radians(lon_ref)), np.sin(np.radians(lat_ref))]
    ])

    # Apply the rotation matrix to the difference vector
    enu = np.dot(R, np.array([dx, dy, dz]))
    
    return enu

def convert_tum_ecef_to_enu(input_file, output_file, lat_ref, lon_ref, alt_ref):
    with open(input_file, 'r') as f_in, open(output_file, 'w') as f_out:
        for line in f_in:
            if line.strip() and not line.startswith('#'):
                parts = line.strip().split()
                timestamp = parts[0]
                x_ecef = float(parts[1])
                y_ecef = float(parts[2])
                z_ecef = float(parts[3])
                qx = parts[4]
                qy = parts[5]
                qz = parts[6]
                qw = parts[7]
                
                enu = ecef_to_enu(lat_ref, lon_ref, alt_ref, x_ecef, y_ecef, z_ecef)
                f_out.write("%s %.6f %.6f %.6f %s %s %s %s\n" % (timestamp, enu[0], enu[1], enu[2], qx, qy, qz, qw))

def get_reference_from_bag(bag_file):
    # Open the .bag file
    with rosbag.Bag(bag_file, 'r') as bag:
        # Iterate through the messages in the bag
        old_type = None
        for topic, msg, t in bag.read_messages(topics='/ublox/fix'):
            #if old_type != type(msg):
	    #print(str(type(msg)))
	    #old_type = type(msg)
            if True:
                # Get the first NavSatFix message and return the reference LLA
                lat_ref = msg.latitude
                lon_ref = msg.longitude
                alt_ref = msg.altitude
                return lat_ref, lon_ref, alt_ref
    return None, None, None  # Return None if no NavSatFix messages are found

if __name__ == "__main__":
    parser = argparse.ArgumentParser(description="Convert ECEF coordinates in TUM format to ENU coordinates.")
    parser.add_argument("input_file", type=str, help="Path to the input TUM file with ECEF coordinates.")
    parser.add_argument("output_file", type=str, help="Path to the output TUM file with ENU coordinates.")
    parser.add_argument("bag_file", type=str, help="Path to the bag file with LLA.")

    args = parser.parse_args()

    lat_ref, lon_ref, alt_ref = get_reference_from_bag(args.bag_file)

    if lat_ref is not None:
        print("Reference LLA: Latitude = {}, Longitude = {}, Altitude = {}".format(lat_ref, lon_ref, alt_ref))
    else:
        print("No NavSatFix messages found in the .bag file.")

    convert_tum_ecef_to_enu(args.input_file, args.output_file, lat_ref, lon_ref, alt_ref)
```